Doodle3D/doodle3d-client #265: disable travel-lines

companje, 2015-01-20:
[image]

companje, 2015-01-20:
Duplicate of #270

peteruithoven, 2015-06-05:
Not really a duplicate if this is issue is for making the traveling on the button lines configurable, instead of just hiding them in the interface.

companje, 2016-01-04:
I think you can achieve this by changing the following line: https://github.com/Doodle3D/doodle3d-client/blob/master/js/gcodeGenerating.js#L163
by:
```
if (enableTraveling && isTraveling) {
```

peteruithoven, 2016-01-04:
But then also checking the configuration, whether it's enabled or disabled on the bottom layers right?

companje, 2016-01-04:
In my opinion the 'travel-lines' are not a feature but more an unwanted behaviour. I know some people like it because it's kind of a poor mans 'raft' because it keeps lose paths connected and helps some printers to stick to the platform. But it was never meant to make it into the release.

peteruithoven, 2016-01-04:
It's has also proven necessary to get prints to stick on the Wanhao printer, that was the main reason to implement it at that time.

companje, 2016-01-20:
implemented since 0.10.9. great!

closed 2016-01-20 by companje