Doodle3D/doodle3d-client #270: Hiding travel-lines

companje, 2015-01-20:
quote Keegan: "The third issue - not really a problem, but it has been a question that I get a lot. If I may refer you to << image3.jpg >>... Many users notice that if you draw discontinuous lines, the programme will automatically add the very thin lines to connect the discontinuous lines such that the print comes out in one piece. I've had multiple requests asking if these thin lines can be omitted. Is there some setting that we can adjust?"

companje, 2015-01-20:
We can add a setting to 'hide' these thin grey lines but they have actually a meaning. They indicate that this is a please where the 3D-printer needs to travel (move from one point to another without extruding filament/plastic). By hiding these lines people won't realise they ideally should make drawings consisting of one continues line.

Currently while printing the first 3 layers travelling is always disabled which causes the lines to be printed as regular lines. This is something we will soon make configurable since it is asked for a lot. I expect this setting to be part of version milestone 0.10.6.

peteruithoven, 2015-06-05:
Since we opened a issue for adding a issue (todo) for making traveling on the button layers configurable this issue would be about making these lines visible in the sketch interface?