Doodle3D/doodle3d-client #273: corrupt wifibox-image.index

companje, 2015-01-24:
in wifibox.log:
```
01-01 00:04:36 (debug)     Downloading file 'http://doodle3d.com/updates/images/wifibox-image.index'
01-01 00:04:37 (info)     received request of type GET for network/status
01-01 00:04:37 (info)     remote IP/port: 192.168.3.61/58691
01-01 00:04:37 (info)     network:getStatus
01-01 00:04:37 (debug)     ignoring unrecognized field in index file '200	2015-01-24 12' (line 85)
01-01 00:04:37 (debug)     Downloading file 'http://doodle3d.com/updates/images/wifibox-image.index'
01-01 00:04:38 (debug)     ignoring unrecognized field in index file '200	2015-01-24 12' (line 85)
01-01 00:04:38 (debug)     Downloading file 'http://doodle3d.com/updates/images/wifibox-image.beta.index'
```

in wifibox-image.index:
```
Version: 0.10.5
Files: doodle3d-wifibox-0.10.5-tl-mr3020-sysupgrade.bin; doodle3d-wifibox-0.10.5-tl-mr3020-factory.bin
FileSize: 3080196; 3932160
MD5: 8953d1ab6feb8cda3b29731d0a63b352; 1d7fac65b099e56133f8a0b1ad3eff0d
ReleaseDate: 20141223
200	2015-01-24 12:23:30	82.157.62.147	wifibox-image.index	Mozilla/5.0 (Macintosh; Intel Mac OS X 10_10_1) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/39.0.2171.95 Safari/537.36
```

there seems to be debug logging added to the downloaded files. Probably for text files as well as for binary images.

companje, 2015-01-24:
I fixed a bug in the download.php script: removed `die($msg)` in msg function.

closed 2015-01-24 by companje

peteruithoven, 2015-01-24:
Could you clarify? 
- What debug logging? 
- What download.php? 
- Didn't we test?