Doodle3D/doodle3d-client #277: wifibox cannot connect using WiFi: 'wlan0' not found

casperlamboo, 2015-04-15:
Suddenly the WiFi connection does not work anymore. Both creating an AccessPoint and joining as a client fail. Connection with ethernet cable works fine.

details:
- firmware version: 10.5
- When creating an accesspoint the wifibox fails to retrieve the MAC address so it tries to create a network with the name Doodle3D-000000
- looking in the ```/tmp/wifibox.log``` shows ```wireless device 'wlan0' not found, trying fallback 'radio0'``` multiple times
- the folder ```/sys/class/net``` contains: ```br-lan  eth0    lo```. **This shows that wlan0 is missing**
- the contents of /etc/config looks normal

casperlamboo, 2015-04-15:
the folder ```/sys/devices/platform/ar933x_wmac``` exists but the subfolder ```net``` is missing.

casperlamboo, 2015-04-15:
reflashing the wifibox (using ```sysupgrade -v -n /tmp/doodle3d-wifibox-0.10.5-tl-mr3020-sysupgrade.bin```) solved the issue. the cause of the problem is still unknown.

arneboon, 2015-04-15:
was it renamed?

check 
```
sudo gedit /etc/udev/rules.d/70-persistent-net.rules
```

check the NAME tag and rename back to wlan0 if it is changed?

arneboon, 2015-04-15:
```
cat /etc/udev/rules.d/70-persistent-net.rules
```

arneboon, 2015-04-15:
at least that works on debian

casperlamboo, 2015-04-15:
/etc/udev/ does not exist on OpenWrt.

casperlamboo, 2015-04-15:
The ```syslog``` in ```wifibox.tgz``` shows the following on the broken device:
```
Thu Jan  1 00:00:17 1970 daemon.info sysinit: 'radio0' is disabled
Thu Jan  1 00:00:17 1970 daemon.info sysinit: 'radio0' is disabled
Thu Jan  1 00:00:17 1970 daemon.info sysinit: 'radio0' is disabled
```

olijf, 2016-04-06:
On recent firmwares it looks like the network name `` Doodle3D-000000`` comes up more often if you issue a reboot from the command line. Maybe this has something in common with this issue?