Doodle3D/doodle3d-client #291: Update to 10.6.0 appear issue

jeroensomers, 2015-07-01:
When we did the update tests, all went perfect and there where no issues found. 
After the release we got some mails from people who can't connect to the WiFi-Box after updating from 10.5.0 to 10.6.0 .

After the update the unsuccessful situation is:
a ) The WiFi-Box does not appear on connect.doodle3d.com
b ) The WiFi-Box does not open a access point

As a result the WiFi-Box cannot be connected to.
Only with a ethernet cable you can connect and make the WiFi-Box join again, this works.

jeroensomers, 2015-07-01:
Below is the support mail from one of our users:

> "Hi,doodle3d
My WIFI-BOX can not connect after upgrading firmware WIFI, Because I can't find the box WIFI signalal,and can not use a network cable,Please help me how to solve this problem , thank you!
my box s/n 2145513002730."

Reply:

> We have been able to reproduce this situation, and we gave the following solution.
1. Power on the WiFi-Box
2. Connect a Ethernet cable between the WiFi-Box and your PC
3. Go to following address connect.doodle3d.com and wait 30 seconds until the Wired WiFi-Box appears.
4. Alternatively you can visit this address: http://192.168.5.1
5. Proceed as normal, join a wireless network or create a hot-spot from the settings menu.

> It is only possible to resolve this situation with an Ethernet cable  (as we know for now).

arneboon, 2015-07-01:
related issue thread https://github.com/Doodle3D/doodle3d-firmware/issues/38

peteruithoven, 2015-07-03:
After the sign-in fails we'd like to open a accesspoint, see: https://github.com/Doodle3D/doodle3d-firmware/issues/41

There is also a small issue the following issue occurs. https://github.com/Doodle3D/doodle3d-firmware/issues/39 (Box connected to WiFi network, but can't use internet?)