Doodle3D/doodle3d-client #293: Heated bed temperature thermometer Icon under the nozzle temperature thermometer.

jeroensomers, 2015-07-22:
I would love the idea of seeing a icon that tells me how hot my heated bed is, cause when I print on the UM2 I don't know how long I must wait until its ready to print. And I guess that counts for all 3D printers with heated bed enabled.

And if this is going to be in it, make sure that the icon will only appear when a printer type with heated bed is selected.

peteruithoven, 2015-07-22:
We never could give this priority this since a heated bed doesn't really work with the regular usage of Doodle3D, where you rapidly print small prints. With a heated bed you'd have to wait longer before and between prints.

peteruithoven, 2015-07-22:
Duplicate of: https://github.com/Doodle3D/doodle3d-client/issues/170

closed 2015-07-22 by peteruithoven