Doodle3D/doodle3d-client #298: Logic changes to make use of gcode buffering improvements

woutgg, 2016-02-05:
With the modifications to print3d's GCodeBuffer ([details here](https://github.com/Doodle3D/print3d/blob/feature/logging%2Bchecking-renewed/src/drivers/GCodeBuffer.cpp#L9), integration in progress), it is now possible for the server to perform several sanity checks while appending code as well as rejecting new code when the buffer is full.

For this to work though, the client should undergo some changes:
* [ ] API docs on the wiki should be updated with the new metadata fields (in requests) as well as new possible errors (in responses);
* [ ] it should send sequence number, total number of of chunks and preferably also a source ID with each GCode part, so it can be guaranteed the GCode is reconstructed correctly on the server side;
* [ ] it should handle errors when sending GCode - this is currently being ignored, see [here](https://github.com/Doodle3D/doodle3d-client/blob/master/js/Printer.js#L183);
* [ ] moreover, it should deal with rejections when the buffer is full, possibly by exposing [getMaxBufferSize](https://github.com/Doodle3D/print3d/blob/feature/logging%2Bchecking-renewed/src/drivers/GCodeBuffer.h#L55) through the API somehow, so the client can fill up the buffer and wait until it drops below a certain percentage (e.g. to 100%, back to 60%, fill back up again, etc.);
* [ ] although not something to be done in the client itself, but with a small change in print3d, accuracy of printing progress could be improved ([details here](https://github.com/Doodle3D/print3d/issues/34)).

**TODO** for this issue: add overview of related issues that could be solved by this one.

peteruithoven, 2016-02-17:
See branch: https://github.com/Doodle3D/doodle3d-client/commits/feature/buffer-full-retry

woutgg, 2016-02-17:
* API documentation changes are in the [printer section](http://www.doodle3d.com/help/api-documentation#printer).
* Handling gcode send failures in the client implemented [here](https://github.com/Doodle3D/doodle3d-client/commit/578fba4f6379dacbd8842d81bce4a85bdc98daea#diff-698188a6ff78c122c64a5d6a861cbfaeR239).
* Handling buffer full failures: changes in the [client](https://github.com/Doodle3D/doodle3d-client/commit/a9f7fe1c78bd543faa2ed692c092f738550176a3), [firmware](https://github.com/Doodle3D/doodle3d-firmware/commit/f3e9ebf26e482d7f8fbd7044f9f105476ca2d62d) and [print3d](https://github.com/Doodle3D/print3d/commit/f9f1421b5e47e03f0bd276f72957e0ff62ebb987).
* An overview of changes for sending total lines can be found in this issue: https://github.com/Doodle3D/print3d/issues/34.
* Sending sequence numbers from the client is implemented [here](https://github.com/Doodle3D/doodle3d-client/commit/578fba4f6379dacbd8842d81bce4a85bdc98daea).