Doodle3D/doodle3d-client #299: Drawing space doesn't correspond with UM2 Go bed dimension

alx, 2016-02-22:
Hi Doodle3d team,

Thanks for your project, I've bought it during kickstarter and tried it for the first time during a few days at an exhibition of a new public/scientific venue in Toulouse, France : http://quaidessavoirs.fr/

During this time, we used an iPad and a UM2 Go, with the correct preset in doodle3d settings.

I needed to put sticker on the iPad screen, because if people tried to draw on the entire screen, then the print was too large for the UM2 Go bed.

Have you got an idea to solve this issue without putting stickers on the screen? What could I try next time to solve this issue?

Thanks and have a nice day,

Alex

companje, 2016-02-22:
Hi Alex,

We're trying to find an elegant solution for this. In the meantime you can
set the "Pixel to mm scale" in the Doodle3D settings to 0,2. It will stay
within the limits then.

Best,
Rick

[image: Inline image 1]

companje, 2016-02-22:
[image]

alx, 2016-02-22:
Thanks a lot!

Feel free to close this issue.

closed 2016-02-22 by companje