Doodle3D/doodle3d-client #300: Can't go to "connect.doodle3d.com" within access point mode (Beta version 0.10.10-a)

jeroensomers, 2016-02-29:
This is what happens after the update to beta version "0.10.10-a" 
I connect to the access point and go to "connect.doodle3d.com" but I can't get access to it..
Also with a Ethernet cable connected I can't get access.

But when I go to "192.168.5.1" it automatically goes to the drawing window and it prints fine.

jeroensomers, 2016-02-29:
[wifibox-logs.zip](https://github.com/Doodle3D/doodle3d-client/files/150998/wifibox-logs.zip)

peteruithoven, 2016-02-29:
Not getting access to the regular connect.doodle3d.com makes sense in this case, but you should have gotten the drawing interface instead. But you where only able to get that by going to the ip address directly.

peteruithoven, 2016-02-29:
Compared to another WiFi-Box with the same firmware I don't see logs that stand out...
What happens after a reboot? 
What happens in client-mode?

companje, 2016-02-29:
I wonder if this can be caused by your browser. This is my theory: May be it doesn't recognise 'connect.doodle3d.com' as a URL. Instead it tries to connect to the HTTPS version of for example Google. The WiFi-Box doesn't run a HTTPS server so it fails.

peteruithoven, 2016-03-03:
So we could try going to a new url and using `http://` explicitly, like: 
http://something.com

jeroensomers, 2016-03-08:
I tried going to the drawing window by using http:// but it didn't work.
tried it in Firefox, chrome and IE.

peteruithoven, 2016-03-08:
Jeroen (on Windows) still had this after a rebooting the WiFi-Box, he has also replicated the issue with another WiFi-Box. (so Doodle3D-4CCE92 and Doodle3D-EC9EB6)
I've tried connecting to the same WiFi-Boxes with the Raspberry Pi  (Rasbian, with Midori and Chrome browser) and my Laptop (Elementary OS with FireFox) and I don't have this problem, I do get the drawing interface. 
Let's try this on another Windows computer.

peteruithoven, 2016-03-08:
A colleague, also on Windows 8.1 doesn't have the issue. Another on Windows 10 has the same issue. On Chrome it mentions: 
```
DNS_PROBE_FINISHED_NO_INTERNET
```
When connected and trying to reach a non existing website it says 
```
ERR_NAME_NOT_RESOLVED
```

Let's double check this is happening since this beta, let's try to connect on Windows to a WiFi-Box with the previous stable release.

peteruithoven, 2016-03-08:
I think this is where we configure the redirection:
https://github.com/Doodle3D/doodle3d-firmware/blob/master/src/network/netconfig.lua#L218
We add a `addres` to the `dnsmasq`, excerpt from box: 
```
config dnsmasq
  ...
  list address '/#/192.168.10.1'
```
Can't find this option in the docs though: https://wiki.openwrt.org/doc/uci/dhcp

Another possibly related config: 
https://github.com/Doodle3D/doodle3d-firmware/blob/master/src/network/netconfig.lua#L255-L263

peteruithoven, 2016-03-08:
The same happens with the latest stable release, which means it was an existing issue. 

Luckily connecting the box to a WiFi-network using connect.doodle3d.com flow works, because it checks the ip address directly.

woutgg, 2016-03-09:
> ```
> config dnsmasq
>   ...
>   list address '/#/192.168.10.1'
> ```
> Can't find this option in the docs though: https://wiki.openwrt.org/doc/uci/dhcp

The address option is documented in the dnsmasq [man page](http://www.thekelleys.org.uk/dnsmasq/docs/dnsmasq-man.html#lbAE) (look for the `-A, --address` option).
The dnsmasq init script applies the UCI flag to the command line ([see here](https://dev.openwrt.org/browser/trunk/package/network/services/dnsmasq/files/dnsmasq.init#L154)), but it seems to be missing from the documentation indeed. Perhaps we should add it?

jeroensomers, 2016-03-15:
I cleaned my whole laptop, fresh install and upgraded to windows 10.
Now when I connect to the access point and go [connect.doodle3d.com](url) it acts normal.
I can type any url I want and it still goes to the Doodle3D drawing screen.

companje, 2016-03-15:
Thanks Jeroen, so this probably means it was caused by a misconfiguration of the network settings on your previous Windows install. I'm closing this issue. Feel free to re-open when necessary.

closed 2016-03-15 by companje