Doodle3D/doodle3d-client #303: Print stops on the UM2 with extra code added trough the web-console

jeroensomers, 2016-03-09:
When testing the Beta version 0.10.10-a on the UM2 I inserted the following code first in the web-console:
```
MAX_POINTS_TO_PRINT = Printer.MAX_GCODE_SIZE = 999999999999999999999;
Printer.MAX_LINES_PER_POST = 2000;
```

Now when I send a huge drawing to the printer it stops in the middle of the print and gives no error in the web-console.

I attached the web-console logs and the WiFi-Box logs.
[webconsole_UM2_logs.txt](https://github.com/Doodle3D/doodle3d-client/files/165707/webconsole_UM2_logs.txt)

[wifibox-logs.zip](https://github.com/Doodle3D/doodle3d-client/files/165708/wifibox-logs.zip)

woutgg, 2016-03-11:
The print server stops printing new lines at some point, from `webconsole_UM2_logs.txt`:
```
progress:  176766/503927 (51233) (printing) doodle3d-client.min.js:2:4970
progress:  176843/503927 (51156) (printing) doodle3d-client.min.js:2:4970
progress:  176926/503927 (51073) (printing) doodle3d-client.min.js:2:4970
progress:  176926/503927 (51073) (printing) doodle3d-client.min.js:2:4970
```

Without the gcode I think this will be hard to debug. Do you think this issue can be reproduced?

peteruithoven, 2016-03-11:
It's probably some wordart, extruded pretty high. 

We probably need to log more when we attempt to reproduce this? Meaning we probably need to fix log rotation first.

peteruithoven, 2016-03-14:
We have a new image (and some separate tweaks) with more logging so we can test this again. Because the change includes logrotation we can (and should) now test with logging in bulk mode.

jeroensomers, 2016-03-15:
These are the logfiles with the new image (10.10.b)
Webconsole logs
WiF-Box logs
logrotation logs

[webconsole logs 10.10.b.txt](https://github.com/Doodle3D/doodle3d-client/files/174293/webconsole.logs.10.10.b.txt)

peteruithoven, 2016-03-15:
This print didn't fail, but there are errors in the print3d logs for some reason. 
We'll try creating print3d logs in bulk mode next.

peteruithoven, 2016-03-15:
A second, very short lasting attempt. It very quickly stopped printing. 

Since we didn't restart some logs might contain logs from the previous attempt. The print3d logs where removed before this attempt, so that contains only this attempt.  
- [wifibox-logs.zip](https://github.com/Doodle3D/doodle3d-client/files/174431/wifibox-logs.zip)
- [webconsole-logs](https://github.com/Doodle3D/doodle3d-client/files/174432/webconsole-log-10.0.0.123-1458060747653.txt)

peteruithoven, 2016-03-16:
The next attempt was succesfully printed. For reference, the logs
- [wifibox-logs.zip](https://github.com/Doodle3D/doodle3d-client/files/175680/wifibox-logs.zip)
- [wifibox-rotated.zip](https://github.com/Doodle3D/doodle3d-client/files/175681/wifibox-rotated.zip)
- [10.0.0.123-1458122284713.txt](https://github.com/Doodle3D/doodle3d-client/files/175682/10.0.0.123-1458122284713.txt)

peteruithoven, 2016-03-18:
In test https://github.com/Doodle3D/doodle3d-client/issues/303#issuecomment-196933805 we noticed that the client didn't send a second print part. 
@jeroensomers could you try sending two regular sized prints? Do this with log level bulk, without reconnecting the printer or refreshing the client. Console hacks not required.

peteruithoven, 2016-03-22:
Jeroen was able to succesfully print 2 files after each other. Logfiles: 
[wifibox-logs.zip](https://github.com/Doodle3D/doodle3d-client/files/183887/wifibox-logs.zip)

woutgg, 2016-03-23:
Just some more thoughts on the failures/errors from earlier: the errors in this issue as well as in #302 so far all seem to originate from the client-side.

In the non-failed print but with errors in the print3d log ([this one](https://github.com/Doodle3D/doodle3d-client/issues/303#issuecomment-196908694)), I cannot explain the broken pipe errors.
```
15-03 07:00:03 [ABSD] (info)   : setState(): 3:idle > 6:stopping
15-03 07:00:03 [CLI ] (error)  : could not send data in client with fd 8 (Broken pipe)
   (... repeated 30 times ...)
15-03 07:00:03 [CLI ] (error)  : could not send data in client with fd 8 (Broken pipe)
15-03 07:00:03 [GCB ] (error)  : append() - sequence numbering error 4; num/ttl/src stats: own=-1/-1/(null), received=42/252/192.168.10.212
15-03 07:00:03 [CLI ] (error)  : could not send data in client with fd 8 (Broken pipe)
15-03 07:00:03 [GCB ] (error)  : append() - sequence numbering error 4; num/ttl/src stats: own=-1/-1/(null), received=3/252/192.168.10.212
15-03 07:00:19 [ABSD] (info)   : setState(): 6:stopping > 3:idle
```
The incorrect sequence numbers afterwards are even more strange: first 42, then 3, while the print server seems to be idle (inferred from the -1 values). More logs might indeed help here.

In [the very short-lasting attempt](https://github.com/Doodle3D/doodle3d-client/issues/303#issuecomment-196933805), only one chunk was sent by the client, as @peteruithoven already mentioned.

At some point I was curious what browser/OS was used to run these tests. For this reason I changed the info/logfiles endpoint to include the HTTP headers which we could include in the future if there are no (security related?) objections. See [here](https://github.com/Doodle3D/doodle3d-firmware/commit/a49e255b9044c7abba35ac0a28861e40a73e0a72).

woutgg, 2016-04-22:
A recap of the attempts in this issue:

1. (09-3) _progress stopped_ - this looks like something that also occurred several times in Doodle3D/WiFi-Box#3
2. (15-3 by Jeroen) _print successful, strange errors in print3d log_ - no leads, corresp. part of wifibox log not available
3. (15-3 by Peter) _client sent only one chunk_ - unclear why, something happened on the client end?
4. (16-3) _print successful_
5. (22-3) _print successful_

Attempts 2 and 3 cannot be explained, hence we cannot to too much with them unless we find something to connect them with or reproduce them - do we want to create new issues for them?

As attempt 1 looks like something that happened several times in Doodle3D/WiFi-Box#3, perhaps we should create a separate issue to investigate that.

peteruithoven, 2016-04-22:
@woutgg regarding attempts 2 and 3, I'm assuming we didn't log as we need, also since we tweaked what we're logging since then. So let's forget those for now. 

I would open a new issue for the issues we found in https://github.com/Doodle3D/WiFi-Box/issues/3 (and attempt 1).

peteruithoven, 2016-05-25:
For the release of the 0.10.10 we've done lot's of tests. I'm closing this issue for now.

closed 2016-05-25 by peteruithoven