Doodle3D/doodle3d-client #308: Nozzle size setting for UM2 with Olson Block

companje, 2016-05-22:
I received a question from a user regarding setting a custom nozzle size. She got an Olson Block for her Ultimaker 2.
First I thought we had a settings for this in the Doodle3D settings screen but this seems not to be the case.

May be nozzle size doesn't affect the amount of extrusion? With a larger nozzle you can just print faster or allow for a thicker 'wall thickness' using the same speed. Right?

peteruithoven, 2016-05-23:
@jeroensomers what do you think? 
I would imagine that bigger nozzle means extruding more material, that is how bigger nozzles probably result in faster printing. 

What this user can already do is increase the layer height.

companje, 2016-05-23:
I think it can potentionally extrude more per unit of time. But it doesn't automatically extrude more if you don't change the other settings.

jeroensomers, 2017-05-03:
I had a mail from someone that also purchased the Olson Block and found out that you only need to scale up the wallthickness - if you put in a "bigger" nozzle in the machine 0,6 or 0,8 or 1.0. just change the wall thickness.
nozzle size 0.4 == wallthickness 0,5
nozzle size 0.6 == wallthickness 0,7
nozzle size 0.8 == wallthickness 0,9
nozzle size 1.0 == wallthickness 1,1

[image]